F5OEO/rpidatv #1: Video Latency

F5OEO, 2016-03-08:
Output bitrate from ffmpeg seems not set at exactly the bitrate causing rpidatv to wait for samples and wrong timecodes.
Ways to correct : 
- Let libmpegts do the transport encoding
- Make a filling "null packet" process to set the right bitrate coming in rpidatv
- Try gstreamer to see if improve

Output actual ffmpeg transport stream over IP is around 100ms latency.

MatteoDalponte, 2016-03-10:
firstly thanks for your wonderful work.
in this days i'm try to reduce the latency. 
thinking that the latency comes from ffmpeg i tried to use driver uv4l setted with h264 encoding and VLC to read /dev/video0 driver and mux video into the videots fifo file.
Now i have 2 seconds of delay, but are still too many, and then I have some problems to set the bitrate because too low cause buffer empty and too mutch cause latency and  problem to see the video.

[image]

this is the code that i used:

rm videots
mkfifo videots

uv4l --driver raspicam --width 640 --height 480 --auto-video_nr --encoding h264 --bitrate 400000  --enable-server off

sudo chrt -a -r -p 99 `pgrep uv4l`

LD_PRELOAD='' cvlc  v4l2:///dev/video1 --no-drop-late-frames --no-skip-frames --live-caching=0 --v4l2-width 640 --v4l2-height 480 --v4l2-chroma h264 --sout-mux-caching 50 --sout-ts-dts-delay 0 --sout '#std{access=file,mux=ts,dst=-}' > videots & 

sudo /home/pi/RpiDATV/RpiDATV videots 500 1 0 62.5 7 &

here some VLC options about ts transcoding:
https://gist.github.com/revolunet/320db93516ec19eb931b

F5OEO, 2016-03-10:
Thanks in interesting in rpidatv and specially fix the video delay !
Using uv4l is OK (I tested it also) but encoding with different ways (v4l2, uv4l, raspivid) has the same latency : around 50 to 100 ms.
The real problem here is the Transport Stream output : it should be a CONSTANT bitrate and as much accurate as possible.
ffmpeg outputs at less bitrate than what we set.
vlc is not able to make a constant transport Stream .

Other transport Stream output way could be
gstreamer
Write a utility with libmpegts Library

To compute the transport Stream bitrate : TsRate = SymbolRate*2*188/204*FEC
In your example : 500000*2*188/204*1/2 = 460 784 bit/s

MatteoDalponte, 2016-03-16:
I think that gstreamer has the usual problem with latency.
In raspberry pi I recorded a ts file from raspicam with gstramer. I can play it into an other pc and it works well, but when I trasmits it with rpidatv the file received by leadvb has problems being reproduced.

$ 	gst-launch-1.0 rpicamsrc bitrate=480784 preview=0 ! video/x-	h264,width=720,height=480,framereate=30,profile=high ! h264parse ! mpegtsmux ! filesink 	location=videots

So I have decided to try gstreamer on my pc, I wrote a ts fifo file with gstreamer and I read it with vlc. The result: 2 seconds of delay....  where  do  these  bits stop ?

$ 	gst-launch -e -v v4l2src device="/dev/video0" ! x264enc bitrate=498 ! mpegtsmux ! filesink 	location=videots

I don't know if transport stream permits to transmit a 0 latency video.
I tried to trasmit directly h264 file generated by raspivid, but rpidatv returns error in SYNK. 
In your opinion is it possible to transmit directly only a video file and to recive it with leadvb?

thank you
Matteo

F5OEO, 2016-08-17:
Migrate to libmpegts instead of ffmpeg to encapsulate ES to TS.
Latency seems constant.
Go inside low level openmax instead of V4L2 driver for H264.

closed 2016-08-17 by F5OEO