F5OEO/rpidatv #9: Dvb-c signal with the raspberry PI

jongerenchaos, 2016-04-05:
I want to use the raspberry as dvb-c transmitter to add extra channels to my standard cable provider (304 MHz QAM: 64 Symbol: 6875 ). Is it possible or can you make it possible to transmit dvb-c channels?

matiaspl, 2016-04-08:
QAM64 (or any QAM in this matter) may be out of scope. For this you would need a hardware solution, i.e. IT9500-based stick.

futaris, 2016-04-10:
A UT-100C is another hardware solution that works under Linux. It is DVB-T though.

jongerenchaos, 2016-04-10:
I know the DVB-T solution. But i want to add some extra channels to my standard dutch tv subscription.  Therefore i need a dvb-c solution. 

Is this technical not realizable with the standard raspberry solution, or are we missing for now some libraries?

matiaspl, 2016-04-10:
I recall UT-100C does DVB-C too. The problem with rpidatv @ DVB-T
frequencies is the channel width - consumer receivers can only go as low as
6 MHz (the lowest defined in the ETSI standard, with 5 MHz in DVB-H). And
from what I read here: http://www.vivadatv.org/viewtopic.php?f=75&t=335 we
max out at 2 MHz. That's why it takes leandvb to recieve rpidatv signals on
lower frequencies..

>

F5OEO, 2016-04-19:
rpidatv is not DVB-C compliant and will not be without extra hardware. For DVB-T see Hides UT100C.

closed 2016-04-19 by F5OEO