F5OEO/rpidatv #10: I try to decode the stream using standard DVB-S receiver but don't work help me please.

virtualrobotix, 2016-04-08:
I try to decode the stream using standard DVB-S receiver ... but when i search the channel the receiver don't found my modulation ... i use the LNA at 9750 I try to check the modulation with HackRF and #SDR amd i see the modulation but i don't have any kind of software for decode DVB-S with HackRF i search online but found nothing ... do you have some suggestions ? If not i order a RTL-SDR ... but i thought that with HackRF i don't need a RTL dongle help me please :) 
Thanks 
Roberto

virtualrobotix, 2016-04-09:
Today i bought the Lafayette DUO Signal it support DVB-S and DVB-T  , but i cannot receive the signal on it ... there are some limit in frequency transmission or could be only a problem with antenna ?  If  i try to trasmit a frequency > 800 mhz raspberry don't produce correct modulation .... is only my iusse ? Do you have some suggestions ?

matiaspl, 2016-04-10:
Stop me if I'm wrong but for dvb-s reception we're exploiting the fact that
the RF output spectrum is full of high amplitude harmonics. The output
frequency cannot be set anywhere near 1,2 GHz due to hardware limits.

F5OEO, 2016-08-17:
Above 250MHZ, rpidatv use harmonics. Higher frequency, less resolution. Have to mention it in readme

closed 2016-08-17 by F5OEO