F5OEO/rpidatv #12: How to start

DustinSnijders, 2016-04-14:
I'm trying to install your software on a clean Raspberry Pie 1B with the latest edition of Rasbian. I'm sure i do something wrong due to the fact that there is no signal-out after these steps:

1. deploy image
2. expand memory via raspi-config
3. these commando's:

sudo apt-get update && 
sudo apt-get install rpi-update && 
sudo rpi-update && 
sudo apt-get install apt-transport-https && 
sudo apt-get install git && 
git clone git://github.com/f5oeo/rpidatv && 
cd rpidatv/src && 
make -j 4

What do i forget?

Cheees,
Dustin

F5OEO, 2016-04-19:
you can then launch ./scripts/gbmenu.sh

F5OEO, 2016-08-17:
Need to write a basic startup on wiki

closed 2016-08-17 by F5OEO