F5OEO/rpidatv #21: The example for a transmission contains an unknown option

zanco, 2016-05-01:
The example:
Example : sudo ./rpidatv -i sample.ts -s 250 -c 1/2 -o RF -f 437.5 -l 
contains the -o option, which is refused by the program 

./rpidatv: invalid option -- 'o'
rpidatv: unknown option `-o'.

Can / should it be omitted ?

Thanks,
Ben

weddy2015, 2016-05-01:
Hello Ben

The -o must be -m 
Example : sudo ./rpidatv -i sample.ts -s 250 -c 1/2 -m RF -f 437.5 -l 

Try if that works

Edwin

zanco, 2016-05-02:
That works better, Edwin. Althoug there is another issue, with the command line started there is almost no output on GPIO12. I use a cable from a computer reset button, one header, and I tested all ports with this 30 cm wire, but I only get a strong output when I connect it to Pin 12 = GPIO 18. In the documentation I read that Pin 12 could be an IQ output, but on Pin 13 there is no signal.

Any thoughts on this ?  running Raspberry Pi V2 model B v 1.1 

Thanks,

Ben

weddy2015, 2016-05-02:
Hello Ben,

gpio 18 is where the raspberry transmit on whit rf selected.
IQ is what i think data and no hf signal (correct me if im wrong)

You can set the rf signal power on max 7 then your command line wil look like this

Example : sudo ./rpidatv -i sample.ts -s 250 -c 1/2 -m RF -f 437.5 -l  -p 7

Here is are the commands

rpidatv -1.3.0
Usage:
rpidatv -i File Input -s Symbolrate -c Fec [-m OutputMode] [-f frequency output]  [-l] [-p Power] [-h]   
    -i            path to Transport File Input    
    -s            SymbolRate in KS (125-4000)    
    -c            Fec : 1/2 or 3/4 or 5/6 or 7/8    
    -m            OutputMode   
        {RF(Modulate QSK in RF need -f option to set frequency)}   
        {IQ(Output QPSK I/Q}   
        {PARALLEL(Output parallel (DTX1,MINIMOD..)}   
    {IQWITHCLK(Output I/Q with CLK (F5LGJ)}   
    {DIGITHIN (Output I/Q for Digithin)}   
    -f        Frequency to output in RF Mode in MHZ   
    -l            loop file input   
    -p        Power on output 1..7   
    -x        GPIO Pin output for I or RF {12,18,40}   
    -y        GPIO Pin output for Q {13,19,41,45}   
    -h            help (print this help).   

i hope it helps you
groeten
Edwin

zanco, 2016-05-03:
Thanks, Edwin, that helps a lot. Could you please edit the readme because
it mentions GPIO12 as output and not PIN12 (GPIO18) and also change the -o
in the example (and as text in the C source perhaps)

That could make it work for more testers ;-)

Kind regards,
Ben - PE2BZ

weddy2015, 2016-05-03:
Hello Ben 

Im not able to make the changes sorry.
Im just a follower of rpidatv who gives a helping hand.

Greetings 
Edwin

F5OEO, 2016-08-17:
Thanks Edwin for support help. Appreciate

closed 2016-08-17 by F5OEO