F5OEO/rpidatv #22: Raspberry Pi 3 not working

ladecadence, 2016-05-25:
It looks rpidatv is buggy in the raspberry pi 3. Using leandvb on my laptop (or any other sdr app, like gqrx), I can see the signal coming and then going flat after a few seconds, then coming again, etc. I don't know if it's something related to the GPIOs on the pi3. I have the last raspbian version, updated as of today.
When I call rpidatv on the console I got this:

sudo ./rpidatv -i ../video/mire.jpg -s 250 -c 1/2 -m RF -f 437.5 
RPIDATV Version May 25 2016 (F5OEO Evariste) on 
 Model 2Tuning on 437.532046 MHZ (harmonic 5)
Nb PCM STEP (<1000):999

TS Bitrate should be 230392 bit/s
1464181813:921022107 : Fulling buffer 
TSrate in byte =28799
Init Filling Memory buffer 0
.End Init Filling Memory buffer 71440
Filling BUffer : EOF
Memavailable 832840/1880000 FreeSlot=60736/64000 Bitrate : 0.000000
SYNC ERROR 
SYNC ERROR 
SYNC ERROR 
SYNC ERROR 
SYNC ERROR 
SYNC ERROR 
...

F5OEO, 2016-05-25:
It is due to cpu frequency which is changed by the kernel.
Try this : in /boot/config.txt add
disable_pvt=1
force_turbo=1

F5OEO, 2016-08-08:
Carefull : input video file should be a Transport stream (.ts). If you want to transmirt a picture (jpeg) , you have to convert it to transport stream first( with ffmpeg for example). You could see in /scripts/a.sh how the picture is transformed in transport stream.
SYNC ERRO means it is not a transport stream.

csarmo, 2016-08-08:
But i also try some of the .ts file from this page: http://www.dododge.net/roku/ts-samples.html but it gave me the same SYNC ERROR. Maybe the symbol rate of the .ts file i selected is too high or is another thing ?

F5OEO, 2016-08-17:
No matter with symbolrate, seems that it is not a 188 sync byte transport stream or with errors (note in the source page). Need to provide a sample transport stream file to be a reference.

F5OEO, 2016-08-25:
Commit : https://github.com/F5OEO/rpidatv/commit/9df92f7df7be2b469a5fd55a70ff72d65a1a9c9b Should now work with Raspberry pi 3 : need Jessie (bye bye Wheezzy)

closed 2016-09-14 by F5OEO

MendoMiranda, 2017-04-20:
Hi Evariste. 
Regarding this issue I can´t have a good MPEG2 Transmission on Pi3. Spectrum is noisy and incorrect.
I got clean transmission in Pi B+ and received clean in a PI3 with leandvb. 
But, inverting puting the Pi3 in transmissions I can´t receive nor fill the cache ... Spectrum is bad like the picture.
Options on config.txt
disable_pvt=1
force_turbo=1
did nothing better. 
What you suggest to put a clean transmission on Pi3 ?
**Note PI3 RF is on Pin 32
But on Pi B+ is on Pin 12.**
Thank you for fantastic
[image]
 work that allows me to learn something about DVB-S.
Regards and look the pictures
Hamradio CT4BB
[image]
[image]

MendoMiranda, 2017-04-22:
Hi all...
PROBLEM SOLVED
I should have installed the **Raspbian LITE** NOT the full RECOMENDED... 
Spectrum is now just like the second picture !... 
Carlos

F5OEO, 2017-04-22:
Thx Carlos, have to warn to install raspbian lite or debug why full raspbian has issue

MendoMiranda, 2017-05-01:
Yes Evariste. I have done it and I´m testing.
Thank you for your work in the new age of Ham Radio.
CT4BB

MendoMiranda, 2017-05-09:
Hi Evariste and Ladecadence.
My conclusions, using **rpidatv** and **leandvb** with transmissions between PiB+ and Pi3 @ 550KSy and 1/2 FEC (Until now, other  don´t work well)
**1 -** PiB+ Trasnsmiting Ok and Pi3 receiving OK. 1024 Cache fills in 3 seconds OK ...
**2 -** Pi3 Transminting low level and with strange  spectrum  and, PiB+ receiving badly with 1024  cache filling in 2 minutes! At the end, only one perfect  freezed frame appears... !
**3-** Changed SD cards between Pis. Problem keeps on Pi3 Transmission . 
Any idea Evariste ? Is the kernel causing this ? Shell I blacklist somthing more ? 
73 CT4BB
[image]

F5OEO, 2017-05-09:
HI Carlos,

Can you give a test with the dev branch ? I made a commit for issue with RF on Raspberry 3.

MendoMiranda, 2017-05-09:
Hi Evariste
 I did't find the commit on git Branch:dev...
Or what you mean test with dev branch ?... I´m not a specialized git or software designer...
Tnx CT4BB

MendoMiranda, 2017-05-13:
More tests done. 
Pis are 6 meter apart and they are in connection.
PiB+ -> Pi3 good image, cache fills quick but some times no...
Pi3 -> Pi B+ cache always fills slowly  but when full, one perfect freeze image appears.
My conclusion is: 
Noise is present in constellation . So, I have to add a passband filter and amplifier for 437 MHz .

73 and Tnx CT4BB

F5OEO, 2017-05-14:
Thx for analysis... Seems that Pi3 has issue regarding your tests.
Is it with dev branch or master branch ?
To use dev branch : git checkout dev

MendoMiranda, 2017-05-14:
Hi Evariste.

I tested both with same results. Cache filling speed varies with Pis positions !!...

I’m sure the problem is noise. I will filter the frequency and amplify it.

I will let you know the results later. 

73

 

CT4BB

Carlos

http://www.ct4bb.com

QTH : IM58Jr

Enviada: 14 de maio de 2017 10:36
Para: F5OEO/rpidatv <rpidatv@noreply.github.com>
Cc: MendoMiranda <ct4bb@ct4bb.com>; Comment <comment@noreply.github.com>