F5OEO/rpidatv #28: make problem

b4zz4, 2016-08-10:
Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux

> cc -Wall -g -O2 -Wno-unused-variable -Wno-unused-function  -o ../bin/rpidatv rpidatv.c rpigpio.c dvbsenco8.s fec100.c mailbox.c rpidma.c -lm -lrt -lpthread 
> rpidatv.c: In function ‘main’:
> rpidatv.c:1285:15: warning: variable ‘TotalByteRead’ set but not used [-Wunused-but-set-variable]
>     static int TotalByteRead;
>                ^
> rpidma.c: In function ‘InitDma’:
> rpidma.c:43:11: warning: assignment from incompatible pointer type
>   virtbase = (uint32_t *)mbox.virt_addr;
>            ^
> rpidma.c: In function ‘mem_phys_to_virt’:
> rpidma.c:65:18: warning: initialization makes integer from pointer without a cast
>   uint32_t result=(uint8_t *)mbox.virt_addr+offset;
>

SopaXorzTaker, 2016-08-12:
These are warnings, so nothing bad, the program compiled correctly. But @F5OEO, why are there so many unused variables and pointer type incompatibility? Please fix them.

b4zz4, 2016-08-12:
Thanks, I thought that they were errors

F5OEO, 2016-08-17:
Warnings are not errors, specially unused variables not removed because of debug.

closed 2016-08-17 by F5OEO