F5OEO/rpidatv #31: I would like to know if someone has tested rpidatv in the Raspberry Pi Zero, does the program work in it ?

csarmo, 2016-08-17:


F5OEO, 2016-08-17:
It should work, have to set up again my raspi zero to confirm !

virtualrobotix, 2016-08-18:
Hi Evariste i check the master code but on Rpi Zero don't work . Waiting your feedback :)

F1FAQ, 2016-08-23:
I have tried loading rpiDATV onto raspi zero and it works !
Note : I have only tried DATV transmitter part (RF) with last raspi cam

virtualrobotix, 2016-08-23:
What OS / Kernel rev are you using in your test ? On what pin you put the antenna ? 
best
Roberto

virtualrobotix, 2016-08-23:
This is my result of uname -a 
Linux raspberrypi 4.4.17+ #904 Tue Aug 16 16:58:22 BST 2016 armv6l GNU/Linux
The pin is connected in the same place where connected the RPI 2 .

F1FAQ, 2016-08-24:
Here is mine : Linux NanoPi 4.4.15+ #897 Tue Jul 12 18:38:58 BST 2016 armv6l GNU/Linux
"RF" pin same as RPI2
Christian

virtualrobotix, 2016-08-24:
Hi Christian ,
on what Frequency ? Can you share the raw of your sd ... That could be good for understand what is the difference. And for have a good starting point now with last update i'm now at 4.4.18 . 
I'm testing some Evariste patch but now the problem still  ... Evariste is great developer so i know that is solve it :) But other support in testing could help ... :)

virtualrobotix, 2016-08-24:
Today finish the test with Evariste , now it work on RPI 0 , too 
Great Job !!! I think that the new revision will be release in the next days :) 
https://www.youtube.com/watch?v=PgLi_ZMqI-c
_

F5OEO, 2016-08-25:
Work with all raspberry pi, but need at least Jessie...

closed 2016-08-26 by csarmo