F5OEO/rpidatv #35: TS bitrate is different of the value put in the command.

csarmo, 2016-09-13:
I was running the rpidatv program several times with several .ts files. The program run without error, i use the SDR# aplication to watch the RF signal generates by the raspberry pi. The problem i have is when i try to demodulate the RF with leandvb. I got the PSK symbol clouds for half a second then the PSK gone. I was looking some of the issues concerning to the leandvb in its official page and i found two of the issues that could be the cause:

* Internal saturation
* Wrong symbol/sample rate

I think my problem is the second one, so i use some .ts with known bitrate, but i noticed that the rpidatv, during execution in the terminal fprint:

TS bitrate should be <bitrate> is lower that the one i setup in the rpidatv command (in the "-s" parameter). I'm using the B+ raspberry. Please help with this, if i wrong about those values, or if there's a real issue with my rpi model.

F5OEO, 2016-09-13:
You can take, any transport stream file : modulation/demodulation should work , decoding video could be too slow or too fast, but that's an other problem.

-s is samplerate, not bitrate !
Bitrate is SampleRate*2*188/204*FEC
That explain your interrogation.

Have you get the latest leandvb ? What is your command line for
leandvb and rpidatv?

csarmo, 2016-09-14:
I use the wget http://www.pabr.org/radio/leandvb/leandvb.cc command, i think this is the latest leandvb.
I use for the transmitter: sudo ./rpidatv -i media.ts -s 250 -c 1/2 -m RF -f 437.5 -l
In the receiver part i use the command mentioned in the leandvb official page: rtl_sdr  -f 437500000  -s 2400000  -  |  ./leandvb  --gui  |  vlc -

F5OEO, 2016-09-14:
Last leandvb is now on https://github.com/pabr/leansdr
You have to set the symbolrate to leandvb
You can inspect the example script : https://github.com/F5OEO/rpidatv/blob/master/scripts/leandvb2video.sh

csarmo, 2016-09-14:
Ok i was on my way to try that but i has had some issues concerning the rtl_sdr library. Any way thanks.

closed 2016-09-14 by csarmo