F5OEO/rpidatv #40: Minor documentation correction

BevDan, 2016-10-21:
Hello Everiste,
I am about to download / install this latest version, I see that you have put in a lot of effort tidying up and improving on this great little project.
Since you are busy working on rpidatv, can I just mention that the documentation "README.MD" has an error, in the modulator section, it shows the example using "[-o OutputMode]" where the help text says "-m" as outputmode. I believe I had read in a previous issue that this had caused some confusion when trying to use the software in a terminal window.

BevDan, 2016-10-21:
Hello Everiste,

I have downloaded the latest version, and after issuing a gbmenu.sh command, my screen partially fills with the photo's and the CQ bubbles, but I cant exit that screen (well don't know how to exit) to make changes to settings.

73
Bevan
VK5BD

F5OEO, 2016-10-21:
You just have to press "enter" and you return back to the menu. Same with camera : you will see the preview on the screen, but pressing "enter" return back.

BevDan, 2016-10-22:
I tried that and can confirm with camera mode it does as expected, however in pattern mode I do believe the menu may have been appearing, but cant see it as it is "behind" the slideshow / bubbles display, should there be a kill tcanim or similar when returning to the menu?

If I use ssh to access the pi from my pc, I am able to start the gbmenu and pressing enter does return me to the menu in the ssh terminal, but the monitor connected to the pi continues to show the slideshow. the slideshow will flash off and back on again if I then start transmitting from the ssh window.

I also found I had to run the gbmenu.sh with sudo so that it could write new values to the "config" file, I am thinking I should run in su mode all the time?

BevDan, 2016-11-09:
I have attempted to fix the two locations of the -o should be -m references in the files rpidatv.c and README.md, hopefully I have gone about doing so correctly and that it is a small help to you.

closed 2016-11-09 by BevDan

F5OEO, 2016-11-09:
Thx, I will add it. You could make a pull request...hope that I didn't do a
commit since !

BevDan, 2016-11-10:
Sorry if I have messed you around with this, still trying to get my head around github and the correct way of going about making changes.
Do I need to do a "Fork" then make changes and finally a pull request to follow proper procedure?