F5OEO/rpidatv #44: Piping avc2ts output to rpidatv input

csarmo, 2016-10-25:
I don't know if you already have done this, since i haven't seen the command in the code. Is it possible to direct piping the acv2ts output to the rpidatv using the | for redirection ? What could be a example of the command if the answer is yes.

F5OEO, 2016-10-25:
rpidatv don't get input from stdin. So passing avc2ts -> rpidatv is done through a pipe. See a.sh script to see example : 
```
sudo $PATHRPI"/rpidatv" -i videots -s $SYMBOLRATE_K -c $FECNUM"/"$FECDEN -f $FREQUENCY_OUT -p $GAIN -m $MODE -x $PIN_I -y $PIN_Q &
$PATHRPI"/avc2ts" -b $BITRATE_VIDEO -m $BITRATE_TS -x $VIDEO_WIDTH -y $VIDEO_HEIGHT -f $VIDEO_FPS -i 100 -o videots -t 3 -p $PIDPMT -s $CHANNEL $OUTPUT_IP   &
```

csarmo, 2016-10-25:
I know both programs and the parameters that need to be setup in both, but what i don't know is what type of pipe use in order to immediate transmit live video from the avc2ts to the rpidatv without storing the .ts file somewhere.

F5OEO, 2016-10-25:
just make a fifo : 
`mkfifo video.ts`
and refere to it on both softwares.. that's it

closed 2016-10-31 by csarmo

csarmo, 2016-10-31:
I was testing avc2ts with rpidatv with my raspberry Pi Zero. I was using the next commands:

##################### PLACING INTO THE BIN FOLDER ##########
cd /home/pi/rpidatv/bin

##################### SURE TO KILL ALL PROCESS ######################

sudo killall avc2ts >/dev/null 2>/dev/null
sudo killall rpidatv >/dev/null 2>/dev/null

#################### CREATE THE FIFO PIPE ###################

sudo rm videots
mkfifo videots

################## RUNNING THE COMMANDS ################

sudo ./avc2ts -b 330000 -m 462000 -x 416 -y 234 -f 40 -i 100  -o videots -t 0 &
sudo ./rpidatv -i videots -s 500 -c 1/2 -f 437.5 -p 3 -m RF &

At the beggining the program just run awesome, but after some minutes the leandvb start showing lots of errors, some of those are: PCR call too late, TS discontinuity, more than 5 seconds late, and more. Looks like the Raspberry Zero is too slow when both programs are running. The video (i use a raspicam of 5 MP) looks good just before the errors. In some tests i could watch the QPSK Symbol clouds, the percentage increasing and then decreasing to zero, and keep doing this; maybe this is related with the ts dicontinuity error. 

If you need some image of this errors, ask me. Let me tell you that those are greats programs.

F5OEO, 2016-10-31:
Seems that your bitrate calculation is not good for the mux bitrate : Bitrate = 2*SR*188/204
So for SR 500 FEC 1/2 : 2 * 500 * 1/2 * 188/204 = 460784 bit/s (instead of 462000).
It means that avc2ts overflow the mux which could block  the process after some time.
Carefull also that even we set a target bitrate for video (-b330000) , there could be some peaks in bitrate. Suggest to decrease the video bitrate.
Is there any reason why you have a framerate of 40 fps instead of standard 25 fps ?