F5OEO/rpidatv #62: Help with setup

lars18th, 2017-06-28:
Hi,

This seems to be a good project. Also it's live as the "dev" branch is active.

However, for newbies is quite difficult to understand it.
So, please can you help me to understand some things?

- My target is a standard DVB-S STB, then the 23cm band is required, right?

- Using a Pi3 version, and receiving the MPEG-TS over the network, what is the maximum useable Symbolrate? This TS is CBR and the Pi3 doesn't need to process it, only modulate the signal.

- And using an upconverter, will be possible to improve the Symbolrate?

Thank you for this interesing project.

lars18th, 2017-07-21:
Hi,

No one?
It's the project dead?

Regards.

davecrump, 2017-07-21:
Hi Lars

I have developed this project further with help from F5OEO.  See https://github.com/BritishAmateurTelevisionClub/rpidatv and https://wiki.batc.tv/The_Portsdown_Transmitter.  You can also post questions on the forum dedicated to the project at http://batc.org.uk/forum/viewforum.php?f=103.

To answer your specific question, 23 cm band is easiest for a set top box.  I use 1255 MHz.  The max useable symbol rate is 6 MS.  An upconverter would not improve the symbol rate (although I'm not sure I understand your question).

73, Dave, G8GKQ

lars18th, 2017-07-21:
Hi @davecrump ,

Good to hear!
Regarding the BritishAmateurTelevisionClub fork... I know. However, this fork doesn't have enabled the ISSUE reporting. So this is the reason to comment here.
And for the BATC forum... I'll check it!

In any case, thank you! Your response clarifies a lot:
- The 1255 MHz frequency is valid to connect a regular STB.
- The MAX symbol rate is 6 MS... so around 8Mbps with minimal FEC, right?
- Upconverter: My question is regarding the max symbol rate... it's the same using the harmonics for the 23 cm band than for a lower frequency?

My objective is run this project (the rpidatv binary) inside a Raspberry Pi 3 and receive in realtime a TS from the network. The TS will be created using some tools in a PC workstation. Is this mode supported?

Thank you!

F5OEO, 2017-07-21:
Hi,

Just to inform you that 6MS is reachable only with an I/Q modulator. If you plan to use RF mode, it is max 1MS.

rpidatv has a IP input mode (listen to a multicast address) , so you can send TS over UDP to Raspberry which then send I/Q to modulator.

73's Evariste

lars18th, 2017-07-21:
Hi @F5OEO ,

Thank you! So more info:

- IP Input mode from a multicast address is implemented (in wich version?)
- With the RF mode only 1MS is the max.? Sure? Then I feel this isn't good for a regular STB, right?

Another question: regarding the I/Q modulator... the problem is the Pi CPU or the PWM output?

Regards.

davecrump, 2017-07-21:
Hi

Sorry, I was thinking of IQ mode.  QPSKRF is very clever, but in tests just now with the latest BATC Version of the software, I could not get it to work at anything over 700KS.  This was at 950 MHz - the lowest a normal STB will tune.  Some STBs are happy at 900KS (when set to 1MS) but many work well at 1MS.  Sky boxes only manage 10MS.

Evariste might have a software build that can manage 1MS in RF mode - I've added too many features to the BATC version and slowed it down!

The BATC software version does not accept multicast (it played havoc with domestic networks) and just accepts an addressed UDP input.  Details of one use here:  https://wiki.batc.tv/Feeding_Video_to_Portsdown_from_vMix .  

Sounds like you need a  local oscillator and an IQ modulator board to do what you want to do.

Regards, Dave

lars18th, 2017-07-21:
Hi @davecrump ,

Thank you! However, I repeat my question: using the Pi3 the limitation factor is the CPU or the PWM output?

If the problem is the CPU, then I can do all processing in a PC and streaming the I/Q samples with the SDRdaemon tool (http://github.com/f4exb/sdrdaemon).

My objective is use the Raspberry as a IQ modulator. You think it will viable for 1MS or more?

davecrump, 2017-07-21:
I think that the limitation is the PWM Output.  The RPi can easily generate a 6MS Transport Stream, but is not designed to modulate it on to a carrier.  Evariste will be able to confirm.

This is the reason that we have designed a separate IQ modulator: https://wiki.batc.tv/Filter_Modulator_build

Dave