F5OEO/rpidatv #63: Getting started for beginners

Somull, 2017-07-23:
Hey, thanks for this it's so exciting!
I was planning to use the raspberry as tv transmitter for a university project but, if I can use raspberry for some stuff, I can't really know how to manage the Tv technical side...
I tried to start following the steps on https://github.com/BritishAmateurTelevisionClub/rpidatv but i dont know how to receive the transmission on a tv.

Do I need just to tune the tv on my frequency or am I ignoring some steps?
I'm attaching a snapshot of my tv settings which dont work...

Is there a guide I can follow?
Many thanks
[image]

davecrump, 2017-07-23:
I wish it was that simple!  This project makes the Raspberry Pi produce DVB-S signals - similar to those produced by TV Satellites.  Also, to get the type of signal that a normal Satellite set-top box can receive, you need an external modulator for the project.  The issue is the symbol rate - the RPi on its own can produce 500KS, but a set-top box needs at least 1000KS and for this you need an external modulator with the RPi.  Hope that explains.  More details here: https://wiki.batc.tv/The_Portsdown_Transmitter

Somull, 2017-07-23:
Thanks for the explanations @davecrump !
So anyway, even with the right setting, it would never be potentially visible by all the Tvs, but just from satellites decoders/receivers, isn't it?

Is there any possibility to have a tv transmission, not necessarily in digital, not necessarily having a HQ transmission...? like a RpiTx but for tv...

Sorry for my inaccurate vocabulary
Many thanks

matiaspl, 2017-07-23:
With SR under 1000 KS it's hit and miss. Your mileage may vary. There are
some receivers used by satellite DX community that used to lock on signals
with even below 500KS (1st generation Dreambox, Technomate, Vantage, some
TBS PC tuners), but most won't even allow you to manually enter a three
digit number.

Matt

bootrino, 2020-04-15:
It's 2020 now - is the Raspberry Pi 4 able to produce the required 1000KS?