F5OEO/rpidatv #70: Question (please forgive me)

Zibri, 2020-02-19:
I have seen many projects like this for raspberry pi.
I was wondering:
a notebook has a lot of gpio pins driven by PCH or other devices.
Most of them are very common.
Would it be possible to modulate any GPIO (like one connected to a led or something not so vital) to produce a very low intensity signal inside the notebook?
Think of it as a sort of NFC...

Zibri, 2020-02-19:
With this (rpidatv) project for example, the very low intensity signal could then be received by an antenna positioned over the notebook...
Think of this post as pure brainstorming...

Ben1701A, 2021-01-19:
I doubt half of the GPIO are accessible, let alone unused. Also, you need to find drivers unless someone got one into a Linux distro.